FreeJoy-Team/FreeJoyWiki #1: Encoder problem

buelldozer, 2020-05-09:
First of all, superb project!
When connecting an encoder as describe in documentation, buttons are pressed every two encoder steps and not every step.

vostrenkov, 2020-05-09:
Hi,
please specify which encoder do you use. Some types of encoders have 12 pulses for 24 detents (for example PEC11R-4220F-S0012)

buelldozer, 2020-05-09:
Hi, I use Alps EC11B encoder (https://www.mouser.fr/datasheet/2/15/alps_alps-s-a0008379064-1-1733314.pdf)

vostrenkov, 2020-05-09:
Due this documentation your encoder has 1 pulse per 2 detents as i mentioned before. If you need 1 pulse per detent you have to use another encoder

buelldozer, 2020-05-09:
OK, thanks for your fast reply!
Could be nice to have the feature to manage rotary encoders with twice detents, because most of low PPR encoders ( like 8) have always triwe detents.

vostrenkov, 2020-05-09:
You mean encoders with 2 or 4 pulses per detention?

buelldozer, 2020-05-09:
yes, at least encoders with 2 pulses per detention, as they are more common

vostrenkov, 2020-05-09:
This is possible

vostrenkov, 2020-06-16:
v1.6.0 added new modes for encoders, so you can select encoder reading resolution

closed 2020-06-16 by vostrenkov