FreeJoy-Team/FreeJoyWiki #2: Support for multiple HID devices for one STM32 module

beastieboy, 2020-05-21:
Maybe you won't find this as an issue but that would be great to have more than one Joystick virtual device after connecting the module. That would simplify configuration in many systems, especially MAME. Encoder pulses could be mapped to virtual mouse movement as well.

vostrenkov, 2020-05-21:
I dont comletely understand how it should be implemented. Shall i split current device on several HID devices or what?

beastieboy, 2020-05-21:
I'm just thinking about USB composite device with two joysticks. Pin mappings could be assigned then to different "joystick device". This prevents accidental changing of HID device sequence when connected and configured in Windows for example. This is an old problem with MAME and multiple physical Windows gamepads. There's another great project (but very old) wich implements that what mentioned: https://github.com/pellepl/arcadehid

vostrenkov, 2020-05-21:
Got it. Not sure it can be implemented soon, but i had some thoughts about composite device for extending maximum axis/buttons number and adding mouse reports

bayerla, 2020-07-12:
I will use FreeJoy for Flight Simulation Prepare3D Homecockpit.
**Very good news: The FightSim used I/O program FSUIPC and LINDA accept the 128 Freejoy Buttons in 1 Joystick. Perfect!**
Every cockpit hobbyist's wet dream: Freejoy creates up to 128 buttons in 1 joystick.
And now it comes: FSUIPC and Linda accept the 128 buttons too! OK, I tried buttons 1-11 and 128, then everything should go in between, I just don't have that many buttons !!!

[https://fsc-ev.de/fsc2020/forum/index.php?thread/20066-freejoy-joystick-mit-black-pill-stm32f103c8t6-64kb/&postID=90447#post90447](url)

beastieboy, 2020-07-30:
Any news on that? Currently I'm experimenting with "arcadehid" project but as mentioned before, the project is now long abandoned.

vostrenkov, 2020-07-30:
I mostly focused on fixing current bugs and problems so there is no work on this feature yet

bayerla, 2020-08-29:
Perfect with MSFS 2020:
Sorry for the sound forget it.
https://youtu.be/CWXN6PiMTUE