FreeJoy-Team/FreeJoyWiki #3: Toggle press time

evolution80, 2020-06-08:
Is it possible to have no toggle press time. I need a latching switch to act as toggle on off but not repeatedly. I used to do this with MMJOY2. In game now with this awesome software FreeJoy it repeatedly activates the toggle.

vostrenkov, 2020-06-08:
Update is coming soon (i guess this week). Buttons logic is completely reworked.
You can get more info in [Hotas/Hosas/Simpit discord](https://discord.gg/szqaJE7) or on patreon page (sponsors button at the top of the page)

vostrenkov, 2020-06-12:
Hello!
Please check if your issue has gone in v1.6.0

evolution80, 2020-06-16:
The latest update is fantastic. Toggle switching issue fully resolved. Thank you so much for this awesome software. I would love to have documentation on how to fully utilise the software, particularly led setting up. 

Do you have user manual on how to use the software more?

Best regards 
Bryan

vostrenkov, 2020-06-17:
Thanks for your feedback! The only documentation we have now is this repo. It has poor english translation somewhere but most of information how to set up all functions can be found here. The most actual guide is placed in russian part of this wiki but i'm also trying to keep english part updated. So if you can not find something please try to use google translator and take a look in russian guide