FreeJoy-Team/FreeJoyWiki #6: (｡･∀･)ﾉﾞHi

Bliss233, 2020-10-08:
Hello, I have some problems. The CTS 288 encoder cannot rotate once to trigger the button. After four rotations,00 10,12,02,00 rotate to the left for four cycles,00 02,12,10,00 rotate to the right for four cycles.I used CTS 288 encoder 1/4, may I ask how to set it to be used as two buttons?
[image]
[image]
[image]

Bliss233, 2020-10-08:
I need to rotate the encoder 4 times to press the 1 button

vostrenkov, 2020-10-08:
Hi, changing encoder resolution on encoders tab should help

Bliss233, 2020-10-08:
[image]
I've tried to change this option, but it still doesn't seem to work. I can't turn the button to respond.
greetings

vostrenkov, 2020-10-08:
Ok, I will check it soon

Bliss233, 2020-10-08:
Ok, look forward to that
greetings