FreeJoy-Team/FreeJoyWiki #7: Other STM32 micros?

virtual812, 2020-10-19:
Hey there, love the project.
Just wondering if i were to build my own PCB are there other options with regard to STM32 choice?

The Controller-Lite looks like a good starting point, no need for the 32,768 Crystal etc...

Just got in the mail today one of these, [WeAct Black Pill V2.0](url=https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0)
This is based on the STM32F411CEU6 and seems to have a 100MHz clock would this work?

vostrenkov, 2020-10-19:
Hi,
this software targeted on STM32F103 MCU line and won't work with STM32F4 or any other. 
If you want to use F4 line you will have to port the code with many changes

SarwarN88, 2020-12-26:
Hi,
I am using Maple mini that is based on STM32F103C8T6 which should work. I flashed the .hex file to my board using ST-Link v2. However, when I connect the device to my pc using usb it doesn't connect. It not even showing that a usb device is connected. Do I need to flash all the 3 files inside the FreeJoy_v1_6_2b3.zip?

closed 2021-05-18 by vostrenkov