FreeJoy-Team/FreeJoyWiki #8: MLX90393-connection.md [English] 3.3v or 5v?

youmert, 2020-11-10:
Hey, I'm trying to get an MLX90393 board just like the one pictured connected up.
I have swapped the MISO and MOSI lines, but hesitate to connect the 3.3v pad on the MLX90393 board up to the 5v source on the processor board for fear of destroying th 90393.
Meanwhile I have connected to 3.3v source instead, and while the processor is getting values from the 90393, they don't seem to change in response to magnet proximity. 

Q: Is connecting to 5v source as shown in MLX90393-connection.md correct?
-Thanks

vostrenkov, 2020-11-11:
Hey, 
You are right, there's a mistake on the picture. MLX90393 supply should be 3.3V.

P.S. Have you cut the CS bridge on the back side of the board? This bridge makes hard CS pull-up that disallow using SPI

youmert, 2020-11-12:
Thank You very much, I'm still struggling with the board but thanks to your diagram I did know to cut that CS line!
I'll keep trying but may just drop to tle5010 if unsuccessful. Cheers!

closed 2020-11-12 by youmert