FreeJoy-Team/FreeJoyWiki #11: дополнительный ввод даных через уарт с блютуз модуля.

batbsv, 2020-11-30:
сделайте возможность подключить блютуз модуль к уарту. идея- для телефона или планшета пишите свой софт с кнопочками, осями, красивыми шкурками. подключаетесь через блютуз телефона (планшета)  к модулю в джойстике, и пользуясь экраном плашета как дополнительным РУД и переключателем различных режимов.
я это у себя на ардуине промикро 32u4 пробую делать. скорость блютуз 115200. рабочий вариант написанный через 
mit appinventor 
[image]
как познакомился со спрайтами начал переделывать интерфейс
[image]
все элементы кликабельны и крутятся. но в mit appinventor  очень все запутанно ,а другого я не умею, пока забросил. надеюсь вы поняли мысль.  можно очень гибко на планшете-телефоне настраивать все горячие кнопки, выводить закладками определенные группы кнопок, для навигации, для режимов двигателя, просто подкидывая раскладку кнопок и осей. и по блютузу запускать в главный джойстик и он уже отправляет в компьютер . вместо  телефона можно соорудить дополнительный руд с блютуз модулем, со своими кнопками осями  и контроллером будет как в обычных джоях на две руки,но вместо кабеля будет блютуз. также можно через уарт вести отладку программы джойстика на лету. подставляя значения переменных и контолируя что джойстик отправляет в систему.

batbsv, 2020-11-30:
собственно в микроконтроллер требуется ввести протокол для общения ядра с уартом. 
например "X=1024", "b1=on", "Y=32768","kbA=press" ,""kbA=release"
естественно не обязательно такие большие команды делать,это для примера.
пропусконй способности 115200 хватает чтобы плавно через уарт изменять положение оси,например газа,или триммеров, там не нужно огромное быстробействие.

vostrenkov, 2020-12-02:
Добрый день!
Такая система вряд ли появится в рамках этого проекта в ближайшем будущем, так как помимо интеграции самого управления по UART и вписывания этого в текущую систему конфигурирования потребуется ещё разработать мобильное приложение, на что у команды сейчас нет компетенций

batbsv, 2020-12-02:
но вы же можете сделать своеобразный API для основных функций- присвоение осям значений, нажатие или отпускание кнопок. просто интерфейсом бы выступал уарт. а приложение обязательно напишут энтузиазисты,когда распробуют удобность.

vostrenkov, 2020-12-02:
Да, это возможно. В целом задача перекликается с задачей по подключению двух контроллеров друг к другу, только вместо второго контроллера здесь bluetooth модуль.
Я не могу обещать что этот функционал будет сделан скоро, так как есть другие приоритетные задачи, но какая-то возможность по UART получать и передавать данные вполне вероятно появится

batbsv, 2020-12-02:
спасибо большое. подскажите, какое программное  обеспечение достаточно для компилирования выложенных исходников freejoy? достаточно ли демо версии сред разработки? пора уходить из ардуино на более серьезные среды.

vostrenkov, 2020-12-02:
Я работаю в Keil uVision v5, с включенной оптимизацией код влезает в ограничение бесплатной версии. Можно собирать с gcc через makefile.
Кто-то помнится смог собрать в platform io одну из ранних версий, надо поискать по issues

batbsv, 2020-12-05:
Sprint-Layout 6 не от крывает файлы MMJoy2_74HC165.lay6 TLE5011.lay6 и  т.д.  Вы в какой версии их делали?
на какртинках в инструкциях не понятны номиналы резисторов, только бледные красные прямоугольники. пожалуйста, добаляйте в тексе описания подключений  номиналы .
для светодиодных выводов можно использовать 16 канальные светодиодные драйвера STP16C596,MBI5026GF,MAX7219CNG,или сдвиговые регистры 74HC595.

vostrenkov, 2020-12-07:
Платы *.lay6 в папке /3rd-party - это файлы плат для SprintLayout6, разработанные сторонними разработчиками (причем некоторые изначально под проект mmjoy). У вас, вероятно, просто переведенная на русский 5я версия SprintLayout, которой автор перевода сам назначил 6й номер. Вот информация по этому поводу:
https://radioskot.ru/forum/30-1282-1
Wiki - это тоже народное творчество, большинство контента в вики было добавлено туда сообществом, я по большей части занимаюсь только редактурой. Кстати, вы тоже вполне можете сделать pull-request в вики с более наглядными схемами подключения, и я добавлю их в основную ветку.
Сдвиговые регистры для светодиодов вероятно когда нибудь будут, давно хочу их добавить, но есть более важные вещи

batbsv, 2020-12-08:
спасибо,  лайот подошел, микрухи я заказал для светодиодов. буду тестировать как придут.

batbsv, 2020-12-12:
Это нормально что SPI_CLK  такой  " многоимпульсный" ? 
[image]

vostrenkov, 2020-12-14:
На такой развёртке особо непонятно. У вас тут видно не сам клок, а множество циклов опроса

closed 2021-05-18 by vostrenkov