FreeJoy-Team/FreeJoyWiki #12: USB device not showing after flashing the .hex file

SarwarN88, 2020-12-26:
Hi again, Thanks for adding MLX90393 to the project.
I am currently using Maple mini that is based on STM32F103C8T6. I flashed the .hex file to my board using ST-Link v2. However, when I connect the device to my pc using usb it doesn't connect. It not even showing that a usb device is connected. Do I need to flash all the 3 files inside the FreeJoy_v1_6_2b3.zip or just the .hex file? When I flash back the original bootloader everything work fine. So the problem is not in the cable or the maple mini device.  
Thanks again

SarwarN88, 2020-12-26:
Just an update. Today I tested an original STM32 (STM32F103C8T6) and flashed the .hex file. The pc recognised the usb device as a joystick. So it look like the Maple mini STM32F103CBT6 is not supported.

vostrenkov, 2020-12-26:
Yes, there are some periphery differences between C8 and CB lines

SarwarN88, 2020-12-26:
I am currently working to configure the device in the configuration software. There are many issues:
1. The A4 to A7 axis are not working when I connect a potentiometer. The A4 to A7 are assigned to Ry, Rz, slider1, slider2 by default in the software and there are no any movements in them.
2. When I use a potentiometer to any pin from A0 to A3 in the software the potentiometer effect X, Y, Z, Rx values with different strength. For example when I connect to pin A2 it change the Z axis but the other free axis also changing slightly. 
3. Most of the buttons are not working. When I connect B12 to GND it turn green but when I connect B12 to Vcc it is not working. currently only B12 to B15 are working and only when connected to GND. 
Please help me with these issues. 
Many thanks

ilopez, 2020-12-26:
@SarwarN88 , The original STM32 you are testing, was that a "blue pill" board? I have two that I've been trying to program for a couple of days now. ST Link programs them and verifies the writes fine, but once it I plug it into USB it does not appear at all.

I have visually verified that R10 is 1.5K Ohms, per http://amitesh-singh.github.io/stm32/2017/05/27/Overcoming-wrong-pullup-in-blue-pill.html , maybe I need to measure it to confirm for sure. 

@vostrenkov Is there documentation for known troubleshooting steps?  One of the steps in the wiki was to erase the memory, but that failed; but write+verify worked fine. 

`14:56:53 : ST-LINK SN : 48FF740671XXXX1267
14:56:53 : V2J29S7
14:56:53 : Connected via SWD.
14:56:53 : SWD Frequency = 1,8 MHz.
14:56:53 : Connection mode : Normal.
14:56:53 : Debug in Low Power mode enabled.
14:56:53 : Device ID:0x410 
14:56:53 : Device flash Size : 64KBytes
14:56:53 : Device family :STM32F10xx Medium-density
14:56:59 : Error occured during flash mass erase!`

`14:59:18 : ST-LINK SN : 48FF7406718XXXXXX461267
14:59:18 : V2J29S7
14:59:18 : Connected via SWD.
14:59:18 : SWD Frequency = 1,8 MHz.
14:59:18 : Connection mode : Normal.
14:59:18 : Debug in Low Power mode enabled.
14:59:18 : Device ID:0x410 
14:59:18 : Device flash Size : 64KBytes
14:59:18 : Device family :STM32F10xx Medium-density
14:59:24 : [FreeJoy_v1_6_2b4.hex] opened successfully.
14:59:24 : [FreeJoy_v1_6_2b4.hex] checksum : 0x009AB24D 
14:59:43 : Memory programmed in 5s and 47ms.
14:59:43 : Verification...OK
14:59:43 : Programmed memory Checksum: 0x009AB24D
14:59:53 : Disconnected from device.`

SarwarN88, 2020-12-27:
> Is there documentation for known troubleshooting steps? One of the steps in the wiki was to erase the memory, but that failed; but write+verify worked fine

That is the problem they mention it in the form and I think its related to the windows. My Blue pill STM32F103C8T6 also failed to connect using ST-LINK software or to erase memory. The solution is to connect 10K resistor from reset pin to ground. After you connected the resistor, in the ST link click erase memory and press ok. As soon as you did that, remove the resistor before the failed massage shows up. If you did that successfully the memory will be force erased. For the compatibility, as I mention above my Maple Mini STM32F103CBT6 was not showing up after flashing the farmwear. so its not supported by the joystick software at the moment.

pipes80-collab, 2021-01-25:
my problem after flashing the hex file was the USB cable used, replacing it with an Amazonbasic the device was detected