FreeJoy-Team/FreeJoyWiki #13: Armgcc app firmware !Don't work

chuoique, 2021-01-04:
I successful when try compiler on MDK-ARM vs arm gcc, but app's firmware run as image follow:
[image]
[image]

arm gcc boot firmware seem work good.
[image]

vostrenkov, 2021-01-04:
Hi, 
both boot and app should be flashed to the MCU for proper work. Boot is flashed starting from 0x8000000 (starting point of program memory) and App is flashed with offset to 0x8002000.
Are you sure you flash your App to the right place?

chuoique, 2021-01-04:
Thanks your repply,
I tried both that flashed with start adrress (boot+app same your comment) use ST-LINK Utility, OR that flashed first: boot use ST-LINK Utility, second: app use "Freejoy Configurator firmware flahser".
I guess, it maybe conflict in.. etc (armgcc's config, startup_stm32f10x_md.s, or on usb init..).

ps. I had modifier a little follow link: http://amitesh-singh.github.io/stm32/2017/05/27/Overcoming-wrong-pullup-in-blue-pill.html and change some startup file, but it didn't worked.

chuoique, 2021-01-05:
Hi, I had update lastest version  gcc-arm-none-eabi at https://developer.arm.com . It run ok!. thanks you!

closed 2021-01-05 by chuoique