FreeJoy-Team/FreeJoyWiki #14: MLX90393 не реагирует на магнит.

batbsv, 2021-01-14:
MLX90393 не реагирует на магнит, дорожку порезал как в инструкции ,куда копать?

[image]

[image]

судя по обмену,микросхема вроде живая. платка такая как в инструкции вики.

[image]

vostrenkov, 2021-01-14:
Если можешь, постучись в дискорд Magners#1207. У меня усть пара бета версий, где возможно оно починено

batbsv, 2021-01-14:
сожалению тебя не находит . я bat_bsv#9985

closed 2021-01-17 by vostrenkov

michalsok, 2021-01-27:
I cant read Russian but I also have a problem connecting MLX90393
I did cut the CS trace, I did swap MOSI and MISO as picture and for my own sake confirmed with the datasheet that in fact this breakout board is mislabeled.
When I probe the SPI bus the commands do not make sense and the response (status byte) from MLX is 00010001 indicating error.
The first clock pulse looks suspicious and out of sync. Also, are you sending the commands (bytes) MSb first?
Were you ever able to test the MLX90393 successfully?
Please help.
Thank you,

Mike.
[image]

vostrenkov, 2021-01-27:
I confirm there possible errors with MLX90303 communication. It will be fixed in 1.7.0 version coning soon