FreeJoy-Team/FreeJoyWiki #17: STM Board with FreeJoy Firmware 1.6.2 fail to enumerate and start correctly in Win10

KartmanCanada, 2021-01-26:
On boot from power up, Win10/USB subsystem will fail to enumerate/start the FreeJoy device... 

I have 3x STM32 boards connected to this system. Unplugging and replugging seems to not help but removing device in Device Manager and "discovering" new hardware seems to allow the driver to startup and enumerate correctly

vostrenkov, 2021-01-26:
yes, it is known issue. gonna fix it on next update

KartmanCanada, 2021-02-01:
That's awesome! I think I'm running 1.6.2b4 now... will the fix be a full
release to 1.6.3 or another 'b' bump? Just curious as I'm keen for fix so
I'm watching!

Any thoughts on ETA? If it helps, I"m happy to take a load and run some
testing!

Cheers.

vostrenkov, 2021-02-01:
It will be even 1.7.0 because of major changes.
I hope it will be released in February

eviltheknivel, 2021-04-26:
has this problem been resolved?

vostrenkov, 2021-04-26:
@KartmanCanada does version 1.7.0 work for you as expected?

vostrenkov, 2021-05-17:
can you please provide photo of the board you are using? marking on the chip make sense

snagov, 2021-05-19:
Hi...
I bought another board  in another shop and now works fine.

**Bad Board** [https://es.aliexpress.com/item/32792513237.html?spm=a2g0s.9042311.0.0.3da263c0eIJdEi](url)
[image]

**Good Board** [https://es.aliexpress.com/item/1005001506683666.html?spm=a2g0s.9042311.0.0.3da263c0eIJdEi](url)
[image]

eviltheknivel, 2021-05-19:
thats funny. for me the CH32F doesnt work, the other one STM32 with CHN Label works fine for me.

vostrenkov, 2021-05-20:
That mean that better to get a clone than a fake. I've been reported that CH32 may work  being slower than original STM32.
However slower is better than fully unfunctional

eviltheknivel, 2021-05-20:
The CH32 works not correct with 4bit encoder, . The USB is sometime going up to horrible high Response Time. Upload the hex file is nearly 30seconds to upload.

tahir51, 2021-06-07:
is there any fix for this yet? I get the "Unknown Device (Code 43)" Error. Tried multiple things that google suggested. Tried reflashing a few times, 2 different computers. Still doesn't work. Stm32 board is the same as above described as "bad board". Do I need to buy new ones?

snagov, 2021-06-07:
Yes...there are multiple falsificated boards, I solved the problem buying
another one

tahir51, 2021-06-07:
can you tell me where did you buy? and is it garantueed it will work? I don't want anymore gambling :)

Emirgkts, 2021-06-10:
I was having the same problem with orginal stm32f103 i am also getting code 43 error.I flashed bootloder.bin to my card and flashed latest version freejoy.bin via freejoy confirugator over usb it worked for me.

tahir51, 2021-06-10:
> I was having the same problem with orginal stm32f103 i am also getting code 43 error.I flashed bootloder.bin to my card and flashed latest version freejoy.bin via freejoy confirugator over usb it worked for me.

you are giving me hope Emir. Can you please explain how to flash bootloader.bin to my card and where to find it?

Emirgkts, 2021-06-12:
> > Ben de orjinal stm32f103'te aynı sorunu yaşıyordum, ayrıca 43 hatası alıyorum. Bootloder.bin'i kartıma flashladım ve freejoy.bin'in en son versiyonunu usb üzerinden freejoy confirugator üzerinden flashladım, bu benim için çalıştı.
> 
> Umut veriyorsun Emir. Lütfen bootloader.bin'i kartıma nasıl flashlayacağımı ve nerede bulacağımı açıklayabilir misiniz?

Use STMicroelectronics flash loader to loading bootloader.bin i not find file i just copy here 
[bootloader.bin.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6641801/bootloader.bin.zip)

Emirgkts, 2021-06-12:
After flashing with fdti uart adapter, you connect pc to pc via usb
 then you flash firmware via FreeJoyQt (in the advanced options menu)
You need to flash FreeJoy_1_7_0b3.bin