FreeJoy-Team/FreeJoyWiki #19: After flashing firmware FreeJoy_1_7_0b1.hex, win10 can't recognize the device

wineeast, 2021-02-27:
The STM32F103C8T6 board had a 64kb flash size. After flashing the FreeJoy_1_7_0b1.hex or bootloader.hex using the stm32 st-lnk, the win10 can not recognize the device.
In the hardware list, the device had a yellow exclamation mark, and the error code 43.
In the detail information tab, the device describation's value is unkown device, device address is USB\VID_0000&PID_0002\5&13A74B18&0&10, the hardware ID is USB\DEVICE_DESCRIPTOR_FAILURE.
I think there should be some thing wrong with the hex file or the wiki page have some step lost in the flashing firmware.

vostrenkov, 2021-02-27:
Hi, have you ever seen your board working with USB?
What is the value of D+ pull-up resistor (R10 if you are using BluePill board)?
What is BOOT0 and BOOT1 configuration?

vostrenkov, 2021-02-27:
Also you can try this one, there are some minor changes in bootloader
[FreeJoy_1_7_0b2.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6054933/FreeJoy_1_7_0b2.zip)

wineeast, 2021-02-28:
Thanks for the quick respose.
I checked the board with the new hex file, it does't work.
when I flash the firmware the BOOT0 setting as 1 and BOOT1 setting as 0. And it's hard to recognize the resistor's value.
But I try another firmware just as a hid mouse or keyboard, the board does not work also.
I will buy another  STM32F103C8T6 board to try again.

vostrenkov, 2021-02-28:
BOOT0 = 1 means that your MCU boots not from program memory but from embedded bootloader sector which is used for flashing firmware via UART.
https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Flashing-firmware-with-USB-UART-converter.md

Setting both BOOT0 = 0 should solve your issue

wineeast, 2021-03-05:
Finally, it is done.
I buy another STM32F103C8T6 board, and the same step to flash firmware. Now, everything is OK.
Thanks a lot!

closed 2021-03-05 by wineeast