FreeJoy-Team/FreeJoyWiki #20: Info potentiometers to 3 volt or 5 volt

pipes80-collab, 2021-03-02:
in the software context of the axis created by an analog potentiometer, what can change powering it at 3.3 volts or 5 volts?

vostrenkov, 2021-03-05:
5V exceeds allowed values for ADC pins, it won't read values above 3.3V and this can damage the chip

pipes80-collab, 2021-03-05:
I was referring to a context where you have the possibility and the support ... sorry if I didn't specify it

ufocia, 2021-03-13:
> I was referring to a context where you have the possibility and the support ... sorry if I didn't specify it

You can use whatever voltage you want for the pots so long as you bring it down to the limits expected by the micro. Just use  voltage dividers and/or clamp diodes to bring it down to whatever limits your micro expects.

pipes80-collab, 2021-03-14:
yes that I know, my question, even if my English sucks was precise to the context of an axis that a potentiometer creates in the software in the form of a brake, accelerator or clutch, if it can make a difference or not or give advantages or not

pipes80-collab, 2021-03-14:
> You can use whatever voltage you want for the pots so long as you bring it down to the limits expected by the micro. Just use voltage dividers and/or clamp diodes to bring it down to whatever limits [...]

yes that I know, my question, even if my English sucks was precise to the context of an axis that a potentiometer creates in the software in the form of a brake, accelerator or clutch, if it can make a difference or not or give advantages or not

pipes80-collab, 2021-03-14:
better to connect for each potentiometer its 3.3 volts and gnd?  or make a bridge and use only 3.3 volts and gnd?

ufocia, 2021-03-15:
> better to connect for each potentiometer its 3.3 volts and gnd? or make a bridge and use only 3.3 volts and gnd?

I'm afraid I still don't understand you question. However, I'm going to take a stab at it. If you mean whether to connect each end of the pot to + and gnd and then read the wiper, or whether to connect only one of the pot to +, with the other end floating, and then read the wiper, then I would say the former would be my preferred way so long as the input port has good impedance and/or the minimum impedance on the pot is fairly high. This way you should get a linear response. In the second scenario, I suspect that by bridge you mean a pulldown. This will not give you a linear response as the pulldown will have a fixed value.

Regarding your previous question, 5V could give you better noise rejection, but it really depends on the ADCs that you are using. I think you're overthinking it. Analog pots are so unstable, 5V or 3.3V won't make that much difference. If you're shooting for precision then work with digital encoders rather than pots.

pipes80-collab, 2021-03-15:
I am modifying the Logitech G29 pedal board to connect it to usb and with the new potentiometers I have problems, connected by soldering because they have the connection oin larger than the originals.  they practically work but the values ​​on the freejoy configuratorqt program remain blocked

ufocia, 2021-03-15:
> I have problems, connected by soldering because they have the connection oin larger than the originals.

What's an "oin"? What do you mean by the values being blocked? If you run the wiper from the pots directly into the input without a pulldown then your reading between high and low ends of the pot will not change very much. You can use a pulldown, whether by tying the other leg of the pot to ground or tying the wiper/input to the ground with an appropriately high value resistance to enhance the reading. The bottom line is that you want to measure some variable ratio of resistances, where the reference resistance is preferably not the input resistance.

However, perhaps you already fried the input ports with 5V, if that's what you tried. That's where voltage dividers and perhaps even clipping diodes would come in on a new micro board (the old one being fried) or at least new inputs (if the input circuitry is for the particular input and not the ADC is all that was fried).

pipes80-collab, 2021-03-15:
I wanted to say to have the connection holes of the potentiometers wider than the original ones

pipes80-collab, 2021-03-15:
the 5 volts connection was to try to understand the difference between the 5 volts and the 3.3 volts.
I have not connected through 5 volts but through 3.3 volts and I have this problem: I connect the pedal board in the way specified above and try the axes, perhaps by pressing the pedal once I have response of the slide on the configured axis, and maybe pressing it remains stationary and not  I have an answer and therefore no scrolling.  trying the new potentiometers individually, disassembled from the pedal board did not give any problems, is it an interference of the potentiometer connected in contact with the pedal iron?

ufocia, 2021-03-15:
Sorry, I still don't understand the issue. I think you have too many possibilities and not enough data.

pipes80-collab, 2021-03-15:
the problem is in reading the axis data created by the potentiometer, so adc, sometimes I can calibrate and slide the axis bar normally while sometimes it remains stationary, so to speak the axis bar that is in configuration axes and that would be the output of the value given by the potentiometer

ufocia, 2021-03-15:
If it is intermittent it sounds like either a connection issue (broken wire, bad soldering, defective pot, ...) or physical coupling issue, e.g. the pot's knob sliding in the pedal.

pipes80-collab, 2021-03-16:
> > Ho dei problemi, collegati tramite saldatura perché hanno la connessione oin più grande degli originali.
> 
> Cos'è un "oin"? Cosa intendi per valori bloccati? Se esegui il tergicristallo dalle pentole direttamente nell'ingresso senza un pulldown, la tua lettura tra le estremità alta e bassa della pentola non cambierà molto. È possibile utilizzare un pulldown, legando l'altra gamba della pentola a terra o legando il tergicristallo / ingresso a terra con una resistenza di valore adeguatamente elevato per migliorare la lettura. La linea di fondo è che si desidera misurare un rapporto variabile di resistenze, dove la resistenza di riferimento preferibilmente non è la resistenza di ingresso.
> 
> Tuttavia, forse hai già fritto le porte di ingresso con 5V, se è quello che hai provato. È lì che i divisori di tensione e forse anche i diodi di clipping entrerebbero su una nuova scheda micro (quella vecchia viene fritta) o almeno su nuovi ingressi (se il circuito di ingresso è per il particolare ingresso e non l'ADC è tutto ciò che è stato fritto).

you wrote that by connecting to the 5 volt pin I could have damaged the board, shouldn't that be a 5 volt output? and if so, the card shouldn't support up to 5 volts the connected external switches? -32-bit-arm-cortex-mcus / stm32-mainstream-mcus / stm32f1-series / stm32f103 / stm32f103c8.html
now i think to have damage the board :-( i have order another on amazon, arceli stm32f103c8t6

ufocia, 2021-03-16:
The 5V goes through the pot and into the ST32 input (ADC pins, which are not 5V tolerant).

If you have intermittent problems, then a fried board ADC is probably not the case.

You could run the pots alone or with plain button switches with 3.3V no problem. I've done it in the past. If there are chips in the legacy controller you are converting, they may require 5V power or they may be happy at 3.3V, I don't know. I ultimately decided to go with 5V because I was building a universal legacy gameport interface and 5V is the spec for the gameport. If there are no chips then just run it off 3.3V and save yourself the pain of voltage dividing/protecting every 5v intolerant input. In any case, I would still pull the connections between the pots and the inputs to ground with a good size resistor.

See if these writeups make it any clearer:
http://www.emcu.eu/000/STM32_5VtolerantIO.pdf
http://fab.cba.mit.edu/classes/863.16/doc/tutorials/FabISP/FabISP_Demystified.html

pipes80-collab, 2021-03-16:
> I 5V passano attraverso il potenziometro e entrano nell'ingresso ST32 (pin ADC, che non tollerano i 5V).
> 
> Se hai problemi intermittenti, probabilmente una scheda / ADC fritta non è il caso.
> 
> Potresti far funzionare le pentole da solo o con semplici interruttori a pulsante con 3,3 V senza problemi. L'ho fatto in passato. Se ci sono chip nel controller legacy che stai convertendo, potrebbero richiedere 5 V di alimentazione o potrebbero essere felici a 3,3 V, non lo so. Alla fine ho deciso di utilizzare 5V perché stavo costruendo un'interfaccia legacy legacy universale e 5V è la specifica per il gameport. Se non ci sono chip, spegnilo da 3,3 V e risparmia il dolore della divisione / protezione della tensione di ogni ingresso intollerante a 5 V. In ogni caso, tirerei comunque a massa i collegamenti tra i potenziometri e gli ingressi con un resistore di buone dimensioni.
> 
> Guarda se questi commenti lo rendono più chiaro:
> [http://www.emcu.eu/000/STM32_5VtolerantIO.pdf ](http://www.emcu.eu/000/STM32_5VtolerantIO.pdf)
> http://fab.cba.mit.edu/classes/863.16/doc/tutorials/FabISP/FabISP_Demystified.html

to the card that will arrive I will connect the two potentiometers (accelerator and brake) each to a ground and 3.3v port on the card, right to do this or I could find myself the same situation on the reading of the axes by the configuratorqt?

pipes80-collab, 2021-03-18:
same problem with new board, connected the pedals I find that by pressing them several times the axis bar freezes and does not move for a few seconds like frizzy

ufocia, 2021-03-18:
Like I mentioned before, you have a lot going on. A little too much to diagnose remotely. If someone like at it physically, they can probably figure it out in 5 minutes or less.

pipes80-collab, 2021-03-18:
This a pic [image]

pipes80-collab, 2021-03-18:
I found that the block only happens with the configuratorqt software under Windows, not even the one under Linux blocks and the other reading software works, see jstest-gtk and also under windows

ufocia, 2021-03-18:
Again, there is a lot going on.

I am almost 100% certain that it has more to do with the physical setup than the software, e.g. I see strands of cable sticking out from the first pot, who knows whether that shorts out or not. Also, did you actually solder the DuPont wires into the Bluepill or did you just stick them into the vias? Unless you figure it out yourself, your best chance is to have someone local clean up the connections, check for shorts/physical interference, etc.

BTW, this is clearly a design choice, but you only need 4 wires running to the board with 2 pedals wired, 3.3V, GND, and 1 pin for each wiper. I would daisy chain the 3.3V and GND from pot to pot. To wire all three pedals you only need 5 wires running to the board.

Good luck!

pipes80-collab, 2021-03-18:
> Di nuovo, stanno succedendo molte cose.
> 
> Sono quasi sicuro al 100% che abbia più a che fare con la configurazione fisica che con il software, ad es. Vedo fili di cavo che sporgono dal primo potenziometro, chissà se si interrompe o meno. Inoltre, hai effettivamente saldato i fili DuPont nel Bluepill o li hai semplicemente attaccati alle vie? A meno che tu non lo capisca da solo, la tua migliore possibilità è che qualcuno locale ripulisca le connessioni, controlli cortocircuiti / interferenze fisiche, ecc.
> 
> A proposito, questa è chiaramente una scelta di design, ma hai solo bisogno di 4 fili che corrono alla scheda con 2 pedali cablati, 3,3 V, GND e 1 pin per ogni tergicristallo. Vorrei collegare a margherita 3,3 V e GND da un potenziometro all'altro. Per collegare tutti e tre i pedali sono necessari solo 5 fili che corrono verso la scheda.
> 
> In bocca al lupo!

I don't think the connections are broken, they are potentiometers with solder tips, I used the cables that have leads for Arduino and development boards, they are already measuring, each tip enters the hole in the hole of the end of the potentiometer, and each potentiometer  has its 3.3v and gnd.  I don't understand why it has problems only with freejoy, tomorrow I try with other software, otherwise I re-edit everything as you told me, with 3.3v and gnd in common between the two pedals

ufocia, 2021-03-19:
If you didn't solder on the ends but rather just plugged in the Dupont cables into the holes, then they are coming loose. The tips of those cables are meant to be squeezed by an appropriate spring counterpart. Even if you're just putting together a prototype and you don't want to solder you need to have solid connections. I use alligator clips, breadboards and pin headers for connections on prototypes. whether you use a bus type architecture for your power and ground or hub and spoke like you did it, it will not, other than incidentally, fix your problem. Your problem is most likely weak and loose connections in a physically active device.

pipes80-collab, 2021-03-19:
No no, I have soldered all connection at the potentiometer and at the stm32 board

pipes80-collab, 2021-03-19:
basically it is not a big problem if it is only at the app level, once you configure it you don't use it anymore...I thanks u for understand to me how this work

Reksotiv, 2021-03-19:
axes do not work only in configuratorQt under Windows? and works in other app, games, Linux configuratorQt, another PC? Only axes don't work or buttons too?

pipes80-collab, 2021-03-19:
practically it happens that they work well even in configuratorqt, the problem is that if you press the pedal, like about ten times, it stops and you have no more response for at least 10/15 seconds, and this happens both in Windows and in Linux.  in basic software, for example Linux jstest-gtk and in Windows devices, the block does not occur.  I tried racing set-up and the accelerator and brake work well

pipes80-collab, 2021-03-19:
now between work and training I can not, but tomorrow as soon as I can I make a video to better understand the problem .. unfortunately my English is poor, even my Italian is not perfect, 😂😂😂

pipes80-collab, 2021-03-19:
this is the video 

https://user-images.githubusercontent.com/56699284/111841521-60152000-88fe-11eb-9b0f-c33319c67656.mp4

ufocia, 2021-03-19:
I hope I'm starting to understand the problem. Are you saying that the axes work fine in whatever game/application you are running, but not consistently in the QT Configurator?

Reksotiv, 2021-03-20:
try this version
[FreeJoyQt-noReportsDelay.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6174833/FreeJoyQt-noReportsDelay.zip)

ufocia, 2021-03-20:
> i cannot repeat this problem, but try this version

On this note, I noticed that one axis I'm using, when moved rapidly, sometimes drops below a minimum that is typically achievable. I wonder if that is a problem with the pot or perhaps some interaction with the ADC. Any ideas?

Reksotiv, 2021-03-20:
> I wonder if that is a problem with the pot or perhaps some interaction with the ADC
i think this is a pot

vostrenkov, 2021-03-20:
@pipes80-collab can you please share us file of the issued config?

pipes80-collab, 2021-03-20:
> > Mi chiedo se questo sia un problema con il piatto o forse qualche interazione con l'ADC
> 
> Penso che questa sia una pentola

Think you is a pot problem? If I use one pot give same problem

Reksotiv, 2021-03-20:
> Think you is a pot problem? If I use one pot give same problem

no

> when moved rapidly, sometimes drops below a minimum that is typically achievable

Its pot problem.

Share your config file and try this https://github.com/FreeJoy-Team/FreeJoyWiki/issues/20#issuecomment-803238096

pipes80-collab, 2021-03-20:
this is a video with FreeJoyQt-noReportsDelay

https://user-images.githubusercontent.com/56699284/111876405-e719d500-899e-11eb-847a-d623a5bd556c.mp4

[FreeJoy v1.7.0.cfg.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6175939/FreeJoy.v1.7.0.cfg.zip)

Reksotiv, 2021-03-21:
Thanks. It happens if USB exchange period set to 1 or 2 ms. Working on a fix

pipes80-collab, 2021-03-21:
> Thanks. It happens if USB exchange period set to 1 or 2 ms. Working on a fix

improved situation as recommended by you but with a value higher than 3 ms, and I changed the values as per photo for usb devices
[image]

Reksotiv, 2021-03-21:
check this firmware 
[FreeJoy.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6177624/FreeJoy.zip)

pipes80-collab, 2021-03-21:
i load this firmware from the app?

Reksotiv, 2021-03-21:
yes

pipes80-collab, 2021-03-21:
no problem with this firmware, only this, see video

https://user-images.githubusercontent.com/56699284/111910049-06c50200-8a60-11eb-9654-418686048a83.mp4

Reksotiv, 2021-03-21:
> only this

what did you mean?

pipes80-collab, 2021-03-21:
in the video you can see that the second axis sometimes arrives at the beginning of the configuration while sometimes it remains slightly stressed

Reksotiv, 2021-03-21:
problem with potentiometer

pipes80-collab, 2021-03-21:
removing the ground from the potentiometers to the pedal this oscillation changes

ufocia, 2021-03-22:
> > only this
> 
> what did you mean?

I think he means that the reading is a little lower at the end, i.e. not at maximum.

ufocia, 2021-03-22:
> removing the ground from the potentiometers to the pedal this oscillation changes

Yes, you are completely changing the electric properties of the circuit. The microcontroller measures the voltage between the wiper and the ground dirt/dust, slight position difference, supply voltage difference, and even temperature can all affect that reading.

You could set the maximum at the lower value, then you'll have a little deadband, but probably not noticeable on pedals.

pipes80-collab, 2021-03-24:
> The 5V goes through the pot and into the ST32 input (ADC pins, which are not 5V tolerant).
>
> If you have intermittent problems, then a fried board/ADC is probably not the case.
>
> You could run the [...]

could you kindly explain to me what you mean by pulling the potentiometers to ground with a good resistance? maybe with some examples ... I thank you infinitely