FreeJoy-Team/FreeJoyWiki #22: TLE5012B wiring

FrancescoTalotta, 2021-03-11:
Hello,

is the wiring diagram for the TLE5012B the same as that of the TLE5011/TLE5010? 

Thank you

vostrenkov, 2021-03-17:
Hey, it is a bit different because it dont requires CLK (GEN) output. Refer to TLE5012B datasheet for the connection:
[image]
Note that TLE5012B sensors are not widely tested with freejoy so there are issues possible

teeshq, 2023-07-30:
Hi im interested to 5012B i see in 1.7 u add support  for TLE5012b. will pcb from 3rd-party/hardware work ir need some change ?

teeshq, 2023-09-05:
hi just tested today i use same pcb/wiring with 5012  work perfect

Czary2, 2025-09-09:
Hi, started new PCB design to run TLE5012B sensor with FreeJoy. Please chack if my schematic is correct. Following the diagram above, do we only need 5 wires to run the sensor ?

[image]