FreeJoy-Team/FreeJoyWiki #26: Help configuring for a simple digital joystick?

melstav, 2021-04-07:
I installed the precompiled release of 1.7.0b3 on a blue pill and am using 1.7.0b2 of the Qt Configurator.

I'm trying to set up a simple digital joystick, and am having difficulty getting the desired behavior. I was hoping someone could tell me what I'm doing wrong.

In many arcade machines and early 8 and 16-bit game consoles, the joystick or d-pad is simply four switches under the hood, with one pin on each tied to ground. Push the joystick up, the up switch is pressed, and the Y axis is instantly at max. Release the stick, no switches are pressed, and the axis is immediately returned to center. This is the behavior I am trying to achieve.

Using the windows game controller properties tool, the closest I can get is pushing the stick up moves the indicator to max after a short delay. Releasing the stick has no change. Pushing the stick down returns to center after a short delay, and if I continue to hold it down, it goes to min after another short delay.

I have four pins for the stick set as button_gnd. In Button Config, they're assigned to logical button IDs as "button normal" and I've checked the disable box so they're not reported as buttons to the host. Under Axes Config, I have buttons assigned to Up and Down for both the X and Y axis. Step Div is set to 1. 

Is there a library of example config files available?

melstav, 2021-04-07:
It occurred to me that it might be helpful to post the actual config file generated by the configurator.

[digital_stick.cfg.txt](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6274834/digital_stick.cfg.txt)

vostrenkov, 2021-04-08:
Hi,
you're not doing it wrong. this firmware is not supposed to be used in arcade machines hardware. In some cases it could but in case of your issue there is no simple way to make this firmware work in desired way
up and down functions are made mostly for trimming analog axes (it makes sense in flight simulators) but it wont work properly for 4-position joystick because it can not track release condition for several buttons at the same time.

But you could try to fake needed behavior using analog inputs (connected to your buttons) and plus/minus fuctions for axes, but it will only work if you have external pull-up or pull-down resistors on your buttons. I pin a config file as an example
[fake_digital_joy.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/6281418/fake_digital_joy.zip)

melstav, 2021-04-09:
Ah.

I did think about using resistors to create an analog axis with three discrete values. As you point out, that would work fine for an arcade machine where you have access to rewire the controls, but that would not be possible in all cases, such as interfacing an Atari 2600 joystick.

From the little time I've spent playing with it, FreeJoy seems like a great system for creating custom control systems for simulators and for interfacing analog controls. While it would work fine for some of the things I'd like to make, I do need the ability to create simple auto-centering digital axes. 

I have some other firmware packages to look at which may do the needful. I'll be keeping an eye on FreeJoy, though. It just isn't a good fit for my current need.

Thank you.

closed 2021-05-17 by vostrenkov