FreeJoy-Team/FreeJoyWiki #30: Joystick device doesn't get recognized by Windows after flashing freejoy firmware

tahir51, 2021-06-07:
I bought 2 [STM32 boards](https://www.aliexpress.com/item/1005001636656122.html?spm=a2g0s.9042311.0.0.ffb74c4doo3a3s) did everything according to the guide and flashed latest firmware via ST-Link v2. Up to this point everything seems fine and no problems. Then when I connect the STM32 board to my PC with the micro USB cable it shows an error (code 43 - USB device not recognized, unknown device/description) 

I googled the error and tried everything google suggested, tried 2 PC's, 3 different USB cable, reflashed multiple times, tried older firmware versions but error still occurs. I can't configure the joystick because it doesn't show as connected. In device manager it shows a an unknown USB device with the yellow exclamation (it says also error while requesting device description). 

In the ST-Link Utility software it shows the device as connected like in the guide and there are no errors while flashing the firmware, so I believe there is no wiring issue. I tried 2 different wet of wires too. The reviews on the Aliexpress shows satisfied customers (96%) and I bought 2, both have the same issue. So I would assume the boards aren't defective. Optically they seem fine. 

Any chances for help?

tahir51, 2021-06-07:
Ok, after some more googling it seems lika problem with the board. Also when I erase the firmware and then connect, it shows the error. So I guess it has nothing to do with the firmware. Will try [this next](http://amitesh-singh.github.io/stm32/2017/05/27/Overcoming-wrong-pullup-in-blue-pill.html)

tahir51, 2021-06-07:
ok after some more googeling I found out there are fake ones and the chip is not really an STM32F104C8 as marked on the chip but a chinese CS32F103C8T6 .  see [here](https://github.com/qmk/qmk_firmware/issues/7204)

so I guess these won't work. Does Freejoy work with the black pill board for example STM32F401CCU6 ?

Reksotiv, 2021-06-07:
> Does Freejoy work with the black pill board for example STM32F401CCU6 ?

No, only STM32F103C8