FreeJoy-Team/FreeJoyWiki #37: [Question] Encoder as replacement to potentiometer

Pheggas, 2021-09-12:
Hello. Currently i'm really into this project and already started with setting up (more like buying stuff) FreeJoy controller. My question is. As the encoder looks pretty similar to potentiometer. Can it technically be the direct replacement to potentiometer in driving wheel. Personally, i really hate that low accuracy, low resolution of potentiometer on cheap driving wheels (Currently i own [SpeedLink Drift OZ](https://www.speedlink.com/DRIFT-O.Z.-Racing-Wheel-PC-black-orange/SL-6695-BKOR-01)). With this controller i can simply smoothen the output data from potentiometer to become clear and not dancing from one position to another due to low resolution. But i can also simply, maybe, swap those two components?

Reksotiv, 2021-09-12:
Hello. FJ supports high-resolution [encoders](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Encoders-connection.md)
[image]
and you can select the encoder as the [axis source.](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Axis-configuration.md)
[image]

Pheggas, 2021-09-12:
Of course. I know FJ supports it but the main question is if i can directly replace potentiometer with encoder in order to get better resolution and accuracy?

vostrenkov, 2021-09-15:
Encoder is a ralative sensor and it dont know its position at startup (center position by default)
If this is not a problem for you can use it easily. But for getting better resolution i'd better use digital sensors based on hall effect