FreeJoy-Team/FreeJoyWiki #38: [Question] Button matrix knowledge confirmation

Pheggas, 2021-09-14:
Hello. In your FreeJoy Wiki i found [this scheme](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/images/K1.jpg). From that image, i understand that there are columns and rows, with which combination (by pressing or button or simply sorting), i can call a button press inside BluePill Board, right?

The next question is if i need to put diode (something like 1N4148) behind each of the buttons? And what is the purpose of putting there the diode?

Another question, what is the purpose of that switches on right side of image i mentioned above? Do i need that too?

After all of these questions, i'll explain my situation. So, i have Speedlink Drift OZ wheel with 21 buttons on it. So i can simulate every button via using that button metrix?

Reksotiv, 2021-09-14:
> From that image, i understand that there are columns and rows, with which combination (by pressing switch or button or simply sorting), i can call a button press inside BluePill Board, right?

yes

> The next question is if i need to put diode (something like 1N4148) behind each of the buttons? And what is the purpose of putting there the diode?

"The diode prevents a condition called “ghosting.” In a keyboard matrix, ghosting means you see non-existent button pushes."
I don't use diodes in my steering wheel and everything works fine.

> Another question, what is the purpose of that switches on right side of image i mentioned above? Do i need that too?

no, this is for a single button example.

> So i can simulate every button via using that button metrix?

yes, you can connect the buttons into a matrix and connect them to FJ.

Pheggas, 2021-09-15:
Thank you. I'll maybe post some more questions in the future.