FreeJoy-Team/FreeJoyWiki #42: Windows 10 Not seeing newly flashed STM32

dan0nator, 2021-11-22:
I have flashed 5 different STM32 blue pill's and after I connect them to the PC they get power. The red LED come on and the Green LED comes on and then seems to go to half strength. Windows does not recognize anything. I have tried this with the current 1.0.7 fw as well as 1.0.6. if I load the original FM i get a blinking green LED. I programmed all of them with a ST-Link V2 with current firmware. I was able to view the chip under a good camera and it looks to be original and not a clone.

dan0nator, 2021-11-23:
i have also flashed it with a UART with the same result.

Reksotiv, 2021-11-23:
1) check your USB cable, it can only be for charging
2) you can check for fake with this app
[stmTest.zip](https://github.com/FreeJoy-Team/FreeJoyWiki/files/7589785/stmTest.zip)

dan0nator, 2021-11-23:
Thank you. That was it. all of my boards are now working. I had no idea the
cables were power only. Thank you for solving that problem.

dan0nator, 2021-11-24:
everything is working now that i have the correct USB cable.

closed 2021-11-24 by dan0nator