FreeJoy-Team/FreeJoyWiki #49: STM32 led

Pgpau, 2022-01-30:
Hi, can you connect the PL9823-F5 leds in series, if you have to follow the diagram in the Wiki or is there another way to connect them?
I am attaching the diagram of the LEDs connected to arduino
Thank you
[image]

Reksotiv, 2022-01-30:
Hi, FJ does not support addressable RGB LED

Pgpau, 2022-01-30:
Ok, what kind of leds can be used?

Pgpau, 2022-01-30:
This led WS2812B or WS2801

pwn-d, 2022-02-15:
both are adressable, you need "dumb" leds, as far as I can see, mono color can be put in a matrix and rgbs need to controlled individually. I hate to say it but check the wiki https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/LED-connection.md