FreeJoy-Team/FreeJoyWiki #54: Black Pill

Jesse-Pinkman-443, 2022-05-11:
Hello, just wondering if it is a better option to use the Black Pill ?

ejaimesd, 2022-06-20:
hello, did the Black Pill work for you?

Jesse-Pinkman-443, 2022-06-20:
> hello, did the Black Pill work for you?

Unfortunately with further research into the project it seems the black pill would not work unless the free joy project has been updated to do so in the last month.

ejaimesd, 2022-06-20:
Okay... To late for me... I just got the Black Pill this morning... Because I previously got an STM32 but the 32K only so it don't fit the ROM of 64K... 
I will try the Black Pill to night! So I will update you!!!

ejaimesd, 2022-06-20:
Hello, I could flash the rom to the black pill, but the computer does not recognize the black pill board... any clue?

Jesse-Pinkman-443, 2022-06-21:
Damn, I read somewhere that the software for the board would have to be completely rebuilt to work on the Black Pill. Don't panic, maybe they update the project for the Black pill, would be nice :)

thecuress, 2022-11-02:
[image]

it's a new stm32 ,it's compatible  tks