FreeJoy-Team/FreeJoyWiki #84: Help needen for  74HC165 PCB

Rehsifmas, 2024-12-17:
[image]

Hi there,
I'm building a 3d printed helicopter collective for a flightsimulator. I bought the shift register PCB as show in the image. But can anybody help me out how to connect buttons to it and how to set it up. Because I tried with no succes

Thanks in advance

Assamita81, 2025-03-12:
Hey, it's a bit late, but do you still need help with this?

Rehsifmas, 2025-03-13:
Nope, solved it.
Thanks for asking

Solhead, 2025-04-21:
Hi, how did you fix it? I'm having the same problem. Can you tell me how you set it up? Thanks.

Assamita81, 2025-04-22:
> Hi, how did you fix it? I'm having the same problem. Can you tell me how you set it up? Thanks.

Hi @Solhead. Do you have problems connecting the buttons to the board? or the board to the bluepill?

Solhead, 2025-04-22:
I'm having trouble connecting the board to the Blu-ray player; it doesn't recognize the keystrokes. The AliExpress seller gave me instructions that don't match the instructions on the Freejoy Wiki. 

[image]
[image]

Assamita81, 2025-04-22:
I assume the above screens are what the seller sent you. How do you have it connected?
So essentially, you need to set 3 pins in Freejoy: Clock, Latch, and Data, and those seem to be labeled in the board. The only thing is that you need to connect the OUT part of the board to the Bluepill. In your pictures above you are indicating the IN side is connected to the board. That is wrong.

Apachekhoo, 2025-06-17:
[image]

Hi, reviving this topic i have similar issues. My question is do i need to ensure all the GND (marked in red) are connected? 

No matter what i tried the shift register doesnt work.

Assamita81, 2025-06-17:
> [image]
> 
> Hi, reviving this topic i have similar issues. My question is do i need to ensure all the GND (marked in red) are connected?
> 
> No matter what i tried the shift register doesnt work.

Those are ground pins, they are already connected between them. You need to connect one leg of each button to gnd, and the other to any of the rest of the pins. Are you connecting the OUT of the shift register to the bluepill? Have you configured it in the SR tab in freejoy?

Apachekhoo, 2025-06-18:
Yes i connect using OUT
Yes I only connected one GND, i tested all the ground pins circled in red are connected
I have set the buttons to 24 as pullup in the SR tab
It still doesnt work….

I connect it to A (A3,4 and 5) series on the board. Do i have to connect to B series instead?

Apachekhoo, 2025-06-19:
Yes i did.

Assamita81, 2025-06-19:
I don't know what else to say. It may be a faulty one. If you have an arduino board laying around, I'd do some basic tests with it.
Another reasonable explanation is that the bluepill itself is the faulty one. The market is flooded with cheap knock offs that don't work very well. In fact, it's very difficult to find original ones, so that's also a very plausible explanation, that the problem is not the shift register, but the bluepill

Apachekhoo, 2025-06-19:
Ok any idea where to get genuine bluepill? I think i got fake ones from aliexpress.Sent from my iPhone

Assamita81, 2025-06-24:
I don't see anything wrong with it, tbh. I'd suggest either test the shift register with a basic arduino sketch and serial print to see if it's faulty, or test a different one if you have more. I see you are building a tedac, so I guess you have more

Assamita81, 2025-06-25:
As I said before, I don't see anything wrong with it. Maybe it's just a faulty board. I haven't used these myself, but I've seen how they're designed. I designed my own boards and had them manufactured, because I wanted to have individual shift registers, instead of packs of 3. You can find individual ones too, but then you need to add the pull up or down resistors externally. The few times I've had issues with them, it was just some pad not making a good connection

Rehsifmas, 2025-06-25:
Did you connect it to a 3.3 Volt connection? The shift rehister doesn't work on volt

Assamita81, 2025-06-25:
> Did you connect it to a 3.3 Volt connection? The shift rehister doesn't work on 5 volt

Not true, they are rated for 2-6V. Though to be honest, it's worth the try.

Assamita81, 2025-06-25:
Although now that I think of, although the chip itself is rated for that voltage range, depending on the resistors value they use, it could impact the readings, so again, it's worth the try